RFD-FHEM/SIGNALduino_TOOL #2: SD_Device_ProtocolList.json

HomeAutoUser, 2019-04-10:
**Dieses Issues dient zum Austausch von Informationen der Daten von SD_Device_ProtocolList.json**

@Ralf9 bitte prüfen auf Funktion bzw. Plausibilität.

```
{"name":"NC-3911-675", "id":"38", "data": [
      {
       "dmsg":"W38#12A2C5D0C", "user":"Harst", "repeat":"4",
       "rmsg":"MU;P0=-235;P1=496;P2=253;P3=-479;P4=-957;P5=743;P6=-720;CP=2;D=010231023232310231010232323102310101023102323232310102323245656565623232310232310231023102323231023101023232310231010102310232323231010232324565656562323231023231023102310232323102310102323231023101010231023232323101023232456565656232323102323102310231023232310231010232323102310101023102323232310102323245656565623232310232310231023102323231023101023232310231010102310232323231010232324565656562323231023231023102310232323102310102323231023101010231023232323101023232456565656232323102323102;O;"
      }
    ]
},
```

diese wird nicht von @sidey79 Version decodiert.

Ralf9, 2019-04-11:
Sie wird bei mir decodiert und ist plausibel
```
2019.04.11 19:08:21.720 4 : sduinoD/msg get raw: MU;P0=-235;P1=496;P2=253;P3=-479;P4=-957;P5=743;P6=-720;CP=2;D=010231023232310231010232323102310101023102323232310102323245656565623232310232310231023102323231023101023232310231010102310232323231010232324565656562323231023231023102310232323102310102323231023101010231023232323101023232456565656232323102323102310231023232310231010232323102310101023102323232310102323245656565623232310232310231023102323231023101023232310231010102310232323231010232324565656562323231023231023102310232323102310102323231023101010231023232323101023232456565656232323102323102;O;
2019.04.11 19:08:21.720 4 : sduinoD: Fingerprint for MU Protocol id 38 -> NC-3911 matches, trying to demodulate.
2019.04.11 19:08:21.720 5 : sduinoD: Starting demodulation (StartStr: 5656 cut Pos 59; Signal: (?:10|23){36,} Pos 4) length_min_max (36..36) length=36
2019.04.11 19:08:21.720 5 : sduinoD: dispatching bits: 000100101010001011000101110100001100
2019.04.11 19:08:21.720 4 : sduinoD: decoded matched MU Protocol id 38 dmsg W38#12A2C5D0C length 36
2019.04.11 19:08:21.721 5 : sduinoD: dispatch W38#12A2C5D0C
2019.04.11 19:08:21.721 4 : sduinoD: SD_WS_Parse protocol 38, rawData 12A2C5D0C
2019.04.11 19:08:21.721 4 : sduinoD: SD_WS_Parse decoded protocol-id 38 (NC-3911), sensor-id 12
2019.04.11 19:08:21.721 1 : sduinoD: SD_WS_Parse UNDEFINED sensor SD_WS_38_T detected, code SD_WS_38_T_2
```

HomeAutoUser, 2019-04-30:
@elektron-bbs 
hast du die Sache schonmal getestet?

Es geht hier drum die Nachricht zu verifizieren. @sidey79 Wie verfahren wir, wenn bei dir was nicht geht?
@Ralf9 , gibt es andere Nachrichten wo du weisst, das diese bei dir gehen aber bei @sidey79 nicht?

- Kurz weil mobil -

sidey79, 2019-04-30:
Wenn es einen Fehler gibt, würde ich vorschlagen, dass wir ein issue öffnen und es analysieren.

elektron-bbs, 2019-04-30:
Bei mir funktioniert das dispatchen der RAWMSG auch.

HomeAutoUser, 2019-05-06:
@Ralf9 @elektron-bbs 
bitte diese Nachricht testen

```
{"name":"ITS-150", "id":"3.1", "data": [
      {
       "dmsg":"i455414", "state":"off", "user":"Stefan#", "comment":"on off (FF F0)",
       "rmsg":"MS;P0=-11440;P1=-1121;P2=-416;P5=309;P6=1017;D=150516251515162516251625162516251515151516251625151;CP=5;SP=0;R=66;"
      },
      {
```
es erfolgt kein dispatch.

elektron-bbs, 2019-05-06:
Bei mir auch nicht.

Ralf9, 2019-05-06:
Bei mir funktioniert es
```
2019.05.06 19:07:18.081 4 : sduinoD/msg get raw: MS;P0=-11440;P1=-1121;P2=-416;P5=309;P6=1017;D=150516251515162516251625162516251515151516251625151;CP=5;SP=0;R=66;
2019.05.06 19:07:18.081 4 : sduinoD: Matched MS Protocol id 3.1 -> itv1_sync40, bitLen=24
2019.05.06 19:07:18.081 5 : sduinoD: dispatching bits: 0 1 0 0 0 1 0 1 0 1 0 1 0 1 0 0 0 0 0 1 0 1 0 0
2019.05.06 19:07:18.081 4 : sduinoD: Decoded MS Protocol id 3.1 dmsg i455414 length 24 RSSI = -41
```
Wenn es bei Euch nicht funktioniert, hat sich vermutlich hier ein Fehler eingeschlichen.
https://github.com/RFD-FHEM/RFFHEM/pull/518/commits/cfdabfb2f143112ee30542eae32509197d81ab5e
Hier hat Sidey recht umfangreiche und für mich recht komplexe Optimierungen in der sub SIGNALduino_Parse_MS vorgenommen.
Da diese für mich zu komplex waren und ich die Optimierungen nicht nachvollziehen konnte, habe ich sie nicht in meine Version übernommen

Ralf9, 2019-05-11:
Funktioniert es, wenn Du im Protokollhash bei der ID 3.1 float auskommentierst?

HomeAutoUser, 2019-06-12:
@Ralf9 sind dieser Informationen

```
{"name":"MD-230R", "id":"91", "data": [
    {
      "dmsg":"P91#80306644B", "comment":"Atlantic's - vibration", "user":"Harst",
      "readings": {"state":"warning"},
      "rmsg":"MU;P0=-817;P1=406;P2=-415;P3=813;P5=-4068;CP=3;D=01232323232323232323010123232323230101232301012323012323230123230123010151012323232323232323230101232323232301012323010123230123232301232301230101510123232323232323232301012323232323010123230101232301232323012323012301015101232323232323232323010123232323;O;"
    }
  ]
},
{"name":"MD-230R", "id":"91.1", "data": [
    {
      "dmsg":"P91#80306644B", "comment":"Atlantic's - vibration", "user":"Harst",
      "readings": {"state":"warning"},
      "rmsg":"MS;P0=-420;P1=809;P2=-835;P3=399;P4=-4080;D=34323010101010101010101232301010101012323010123230101230101012301012301232;CP=3;SP=4;s=64;e;m0;"
    }
  ]
},
```

nicht von einem MD_2018R?

Nach meinen Aufzeichnungen von dem User gibt es diese 3 Typen:
"door/windows switch MD-210R | Vibration Schock Sensor MD-2018R | GasSensor MD-2003R"

Wenn ja, würde ich den Namen anpassen.

Ralf9, 2019-06-12:
weiß ich nicht, ich habe es von hier  
https://wiki.fhem.de/wiki/Signalduino_Rawdaten#MD-230R

HomeAutoUser, 2019-06-12:
Ja es ist der 2018R. Ich habe nun nochmal das Forum durchschaut und den Wikieintrag. 
Hier wäre die Angepasste JSON mit erweiterung des Switch

```
{"name":"MD-2018R", "id":"91", "data": [
    {
      "dmsg":"P91#80306644B", "comment":"Atlantic's - vibration (ID 63 additionally)", "user":"Harst",
      "internals": {"DEF":"MD_2018R 803066", "NAME":"MD_2018R_803066"},
      "readings": {"state":"warning", "batteryState":"ok", "contact":"Alarm", "deviceTyp":"vibration", "keepalive":"event"},
      "rmsg":"MU;P0=-817;P1=406;P2=-415;P3=813;P5=-4068;CP=3;D=01232323232323232323010123232323230101232301012323012323230123230123010151012323232323232323230101232323232301012323010123230123232301232301230101510123232323232323232301012323232323010123230101232301232323012323012301015101232323232323232323010123232323;O;"
    }
  ]
},
{"name":"MD-210R", "id":"91", "data": [
    {
      "dmsg":"P91#91DF4F4EB", "comment":"Atlantic's - switch", "user":"Harst",
      "internals": {"DEF":"MD_210R 91DF4F", "NAME":"MD_210R_91DF4F"},
      "readings": {"state":"warning", "batteryState":"ok", "contact":"open", "contact_extern":"closed", "deviceTyp":"switch", "keepalive":"event", "sabotage":"closed"},
      "rmsg":"MU;P1=392;P2=-416;P3=804;P4=-822;P5=-4029;P6=-32001;D=151412323412323234141412341414141412341232341414141234123234141412341234141660;CP=1;R=0;"
    }
  ]
},
{"name":"MD-2018R", "id":"91.1", "data": [
    {
      "dmsg":"P91#80306644B", "comment":"Atlantic's - vibration", "user":"Harst",
      "internals": {"DEF":"MD_2018R 803066", "NAME":"MD_2018R_803066"},
      "readings": {"state":"warning", "batteryState":"ok", "contact":"Alarm", "deviceTyp":"vibration", "keepalive":"event"},
      "rmsg":"MS;P0=-420;P1=809;P2=-835;P3=399;P4=-4080;D=34323010101010101010101232301010101012323010123230101230101012301012301232;CP=3;SP=4;s=64;e;m0;"
    }
  ]
},
{"name":"MD-210R", "id":"91.1", "data": [
    {
      "dmsg":"P91#91DF4F4EB", "comment":"Atlantic's - switch", "user":"Harst",
      "internals": {"DEF":"MD_210R 91DF4F", "NAME":"MD_210R_91DF4F"},
      "readings": {"state":"warning", "batteryState":"ok", "contact":"open", "contact_extern":"closed", "deviceTyp":"switch", "keepalive":"event", "sabotage":"closed"},
      "rmsg":"MS;P0=-407;P1=402;P2=-4023;P3=-817;P4=806;D=12131040431040404313131043131313131043104043131313104310404313131043104313;CP=1;SP=2;R=0;m1;"
    }
  ]
},
```

HomeAutoUser, 2019-06-18:
Hallo @Ralf9 

```
{"name":"WH3080", "id":"9", "data": [
    {
      "dmsg":"P9#FA3C1BD4400000CA50051", "comment":"reconstructed bit 1 (ID 42 additionally)", "user":"Ralf9",
      "readings": {"state":"must be added"},
      "rmsg":"MU;P0=2120;P1=-5736;P2=496;P3=-1024;P4=1467;CP=4;R=16;D=0123232323234323434343232323234343434343232343232323234323432343434323434343434343434343434343434343434343434343432323434323432343432343234343434343434343432343234343432;e;"
    }
  ]
},
```

kann es sein, das die RAWMSG der ID nicht richtig ist, weil diese zu 

> Unknown code P9#FA3C1BD4400000CA50051, help me!

führt?

Wie wäre es, wenn wir diese

 `MU;P0=-1424;P1=1417;P2=-1058;P3=453;P4=-24774;P6=288;P7=-788;D=01212121232343232323232323232123232323232121232121212123212121232123212321232121212123212121232321232321212121232323212321212121212121212323467323232323232323212323232323212123212121212321212123212321232123212121212321212123232123232121212123232321232121;CP=3;R=247;O;`

dafür nutzen?

Ralf9, 2019-06-19:
Ja, diese Nachricht kannst Du auch nehmen.
Du kannst auch diese von killah78 nehmen:
https://forum.fhem.de/index.php/topic,67587.msg948851.html#msg948851

Ralf9, 2019-06-20:
Die Nachrichten der WH1080 und WH3080 haben eine preamble.
FFA... sind Temp, Wind und Rain
Bei der WH3080 gibt es noch
FF7...  sind UV und Lux
Bei nicht so gutem Empfang können am Anfang einige Einsen fehlen.