RFD-FHEM/SIGNALduino_TOOL #3: new function - Github_device_documentation_for_README

HomeAutoUser, 2019-04-15:
Einfach mal zur Diskussion freigegeben. Mir schwebt vor, aus der getesteten JSON Datei die Vorlage für die Github README.md anzufertigen. Somit haben wir eine Copy &  Paste Dokumentation. Getestet habe ich es schon und dies würde vieles erleichtern.

Wir müssen uns nur mit den Namen "beschäftigen" und dann diese ggf mal verbessert oder wie eine Festlegung treffen was wir dort erfassen.

HomeAutoUser, 2019-04-16:
So kann eine generierte Liste aussehen. Es erleichtert natürlich die Aktualität und es kommen nur Sensoren hinzu, welche wir getestet haben und auch Nachrichten besitzen.

Supported Devices / Protocols
======
| Device | FHEM - clientmodule | Typ|
| ------------- | ------------- | ------------- |
| ABS700 | CUL_TCM97001 | Weather sensors |
| Auriol IAN 283582 | SD_WS | Weather sensors |
| Auriol Z31743B | CUL_TCM97001 | Weather sensors |
| Chilitec_22640 | SD_UT | Remote controls / wall buttons |
| Clarus_Switch | IT | Remote controls |
| Conrad S522 | SD_WS | Weather sensors |
| Dooya | Dooya | Shutters / awnings motors |
| EAS800z | SD_WS07 | Weather sensors |
| ET-733 | SD_WS_Maverick | Food thermometer |
| FS20 FHT80b | FHT | Heating control |
| FS20 FHT80TF-2 | CUL_FHTTK | Door / window contact |
| FS20 S4A-2 | FS20 | Remote controls / wall buttons |
| GT-WT-02 | CUL_TCM97001 | Weather sensors |
| Hama TS33C | Hideki | Weather sensors |
| Hama TS34A | SD_WS07 | Weather sensors |
| Hama TS36E | SD_WS07 | Weather sensors |
| HE800 | IT | Remote controls |
| ITS-150 | IT | Remote controls |
| JaroLift TDRCT 04W | SD_Keeloq | Remote controls with KeeLoq encoding |
| KL-RF01 | SD_UT | Remote controls / wall buttons |
| MANAX MX-RCS250 | SD_UT | Remote controls / wall buttons |
| MD-230R | SD_UT | Remote controls / wall buttons |
| MD-230R | SD_UT | Remote controls / wall buttons |
| Mebus | CUL_TCM97001 | Weather sensors |
| minify Funksteckdose |  |  |
| NC-3911-675 | SD_WS | Weather sensors |
| Opus_XT300 | SD_WS | Weather sensors |
| Pollin ISOTRONIC |  |  |
| Pollin_551227 | SD_BELL | Door Bells |
| Prologue | CUL_TCM97001 | Weather sensors |
| PV-8644 | SD_WS | Weather sensors |
| QUIGG_GT-7000 | SD_UT | Remote controls / wall buttons |
| QUIGG_GT-9000 |  |  |
| Remote control Sender 7053 | CUL_TX | Weather sensors |
| Remote control Sender 7054 | CUL_TX | Weather sensors |
| renkforce E0001PA | SD_WS | Weather sensors |
| SA-434-1 mini | SD_UT | Remote controls / wall buttons |
| Sempre 92596/65395 | SD_WS | Weather sensors |
| Somfy RTS | SOMFY | Shutters / awnings motors / doors |
| Techmar | SD_UT | Remote controls / wall buttons |
| Techmar | SD_UT | Remote controls / wall buttons |
| Techmar | SD_UT | Remote controls / wall buttons |
| Tedsen_SKXxxx | SD_UT | Remote controls / wall buttons |
| TFA 30.3176 | Hideki | Weather sensors |
| TFA 30.3208.0 | SD_WS | Weather sensors |
| TFA 30.3221.02, Temp | SD_WS | Weather sensors |
| TFA 30.3222.02, Temp | SD_WS | Weather sensors |
| TFA 30.3222.02, Wind | SD_WS | Weather sensors |
| THGR228N | OREGON | Weather sensors |
| THGR810 | OREGON | Weather sensors |
| THR128 | OREGON | Weather sensors |
| Ventus W044 | CUL_TCM97001 | Weather sensors |
| Ventus W132 | CUL_TCM97001 | Weather sensors |
| Ventus W174 | CUL_TCM97001 | Weather sensors |
| visivon remote |  |  |
| Waeco MA650-TX | SD_Keeloq | Remote controls with KeeLoq encoding |
| WH3080 | SD_WS09 | Weather sensors |
| WS2000/WS7000 Serie - AS2000 | CUL_WS | Weather sensors |
| X10 Protocol | RFXX10REC | RFXCOM-Receiver |
| YET100-1 | IT | Remote controls |

sidey79, 2019-04-17:
Sieht gut aus

HomeAutoUser, 2019-04-17:
Funktion ist eingebunden.

Wenn ich nun das erarbeitete Wissen aus der pm Datei https://github.com/RFD-FHEM/RFFHEM/pull/564 mit "umsetze wo wir plausible Nachrichten haben" so könnten wir demnächst die Übersicht aktualisieren für die User um bezüglich der dev-r34 Auskunft zu erteilen welche Devices verarbeitet werden.

closed 2020-12-29 by HomeAutoUser