RFD-FHEM/SIGNALduino_TOOL #4: SD_Device_ProtocolList.json

elektron-bbs, 2019-04-26:
Sollten wir diese Datei nicht auch mit in die controls_SD_TOOL.txt aufnehmen?

Allerdings bemerke ich gerade, das die Listen im Verzeichnis /SD_TOOL beim Update nicht aktualisiert werden. Das funktioniert dann bei der *.json wahrscheinlich auch nicht.
Ein ähnliches Problem hatte @sidey79 ja auch schon bei der Umstellung der SD_ProtocolData.hash.

HomeAutoUser, 2019-04-30:
@elektron-bbs 
dafür ist es notwendig, das man sich einigt.

Ich würde es mit in das update aufnehmen aber selbst dafür sollten wir versuchen eine Spur zu verfolgen und nicht 2 Prohekte auf dem Boden stampfen. ;-)

- Kurz weil mobil -

elektron-bbs, 2019-04-30:
Ich muss mich übrigens korrigieren: Die Listen im Textformat werden doch mit dem Update aktualisiert. Sie landen allerdings im Verzeichnis
\\RASPBERRYPI3\fhem\FHEM\SD_TOOL
und nicht wie wir es gewohnt waren
\\RASPBERRYPI3\fhem\SD_TOOL
Das hatte ich übersehen.
Dann müsste es mit der Datei SD_Device_ProtocolList.json eigentlich auch funktionieren.

Die Updatefunktion ist ja erst mal unabhängig von der Einigung. Solange wir die JSON-Liste alle händisch bearbeiten, bleibt das Format ja wie von @Ralf9 gewünscht.

sidey79, 2019-05-04:
Ich hab mal die controls Datei angelegt (damit sie auch aktualisiert werden kann) und auch das JSON File mit in die Liste der Dateien aufgenommen.

closed 2019-05-11 by elektron-bbs