RFD-FHEM/SIGNALduino_TOOL #6: Beschreibung der Funktionsweise

sidey79, 2019-05-11:
Ich stelle bei diversen Diskussionen immer wieder fest, dass mit die Funktionsweise und auch der Zweck diverser Features nicht klar ist.

Ich habe da jetzt auch mal wieder ein bisschen was rumprobiert und diverse Dateien erzeugt, aber den roten Faden habe ich irgendwie nicht gefunden.

Wäre toll, wenn das mal zum nachlesen aufgeschrieben wird. Als readme oder auch in einem WIKI in Github.

HomeAutoUser, 2019-05-12:
Hallo

>Ich habe da jetzt auch mal wieder ein bisschen was rumprobiert und diverse Dateien erzeugt, aber den roten Faden habe ich irgendwie nicht gefunden.

- Definiere mit mal bitte was du meinst mit keinen roten Faden gefunden?
- "Zweck diverser Features nicht klar ist." ? Da musst du auch wenn genauer werden?

Beschreibung Funktionsweise? Die Aussage weiß ich nicht zu bewerten, weil die Feature sind derzeit alle in der Commandref versehen.

Das gesamte TOOL Paket was erstellt wurde und sehr hilfreich bei diversen Tests, Hilfefunktionen wo man sonst manuell immer den Befehl mühsam zusammen tragen müsste bis hin zu einer Erleichterung beim erstellen einer Dokumentation auf vielen Ebenen.

sidey79, 2019-05-12:
OK. Es werden Dateien gelesen und auch welche erstellt.
Das erfolgt irgendwie mit Set / get Befehlen die in fhemwrb nicht angeboten werden.

Welche Daten werden eingelesen, welche werden jeweils erzeugt und was ist der Zweck?

HomeAutoUser, 2019-05-12:
>Welche Daten werden eingelesen, welche werden jeweils erzeugt und was ist der Zweck?

Ich weiß nicht genau wie ich dir richtig antworte ;-) Sorry, sonst einfach weiter fragen und die Commandref muss ergänzt werden.

**SET**

    Dispatch_DMSG - eine fertige DMSG vom Modul welche dispatch werden soll (ohne SIGNALduino Verarbeitung!)
     → Beispiel: W51#087A4DB973

    Dispatch_RAWMSG - eine Roh-Nachricht welche einzeln dispatch werden soll
     → Beispiel: MS;P0=-16046;P1=552;P2=-1039;P3=983;P5=-7907;P6=-1841;P7=-4129;D=15161716171616171617171617171617161716161616103232;CP=1;SP=5;

    Dispatch_RAWMSG_last - Dispatch die zu letzt dispatchte Roh-Nachricht

    <modulname> - Dispatch eine Nachricht des ausgewählten Moduls aus dem Attribut DispatchModule.

    ProtocolList_save_to_file - speichert die Sensorinformationen als JSON Datei (derzeit als SD_ProtocolListTEST.json)
     → erst nach erfolgreichen laden einer JSON Datei erscheint diese Option

    START - startet die Schleife zum automatischen dispatchen

    Send_RAWMSG - sendet eine MU | MS | MC Nachricht direkt über den angegebenen Sender (Attribut Sendename ist notwendig!)
     → Beispiel: MS;P0=-16046;P1=552;P2=-1039;P3=983;P5=-7907;P6=-1841;P7=-4129;D=15161716171616171617171617171617161716161616103232;CP=1;SP=5;

**GET**

    All_ClockPulse - berechnet den Durchschnitt des ClockPulse aus der Input_Datei

    All_SyncPulse - berechnet den Durchschnitt des SyncPulse aus der Input_Datei

    ProtocolList_from_file_SD_Device_ProtocolList.json - läd die Informationen aus der Datei SD_Device_ProtocolList.json in den Speicher

    ProtocolList_from_file_SD_ProtocolData.pm - eine Übersicht der RAWMSG´s | Zustände und Module direkt aus der Protokolldatei welche in die SD_ProtocolList.json Datei geschrieben werden.

    Durration_of_Message - ermittelt die Gesamtdauer einer Send_RAWMSG oder READredu_RAWMSG
     → Beispiel 1: SR;R=3;P0=1520;P1=-400;P2=400;P3=-4000;P4=-800;P5=800;P6=-16000;D=0121212121212121212121212123242424516;
     → Beispiel 2: MS;P0=-16046;P1=552;P2=-1039;P3=983;P5=-7907;P6=-1841;P7=-4129;D=15161716171616171617171617171617161716161616103232;CP=1;SP=5;O;

    FilterFile - erstellt eine Datei mit den gefilterten Werten
     → eine Vorauswahl von Suchbegriffen via Checkbox ist möglich
     → die Checkbox Auswahl -ONLY_DATA- filtert nur die Suchdaten einzel aus jeder Zeile anstatt die komplette Zeile mit den gesuchten Daten
     → eingegebene Texte im Textfeld welche mit Komma , getrennt werden, werden ODER verknüpft und ein Text mit Leerzeichen wird als ganzes Argument gesucht

    Github_device_documentation_for_README - erstellt eine txt-Datei welche in Github zur Dokumentation eingearbeitet werden kann.

    InputFile_doublePulse - sucht nach doppelten Pulsen im Datenteil der einzelnen Nachrichten innerhalb der Input_Datei und filtert diese in die Export_Datei. Je nach Größe der Datei kann es eine Weile dauern.

    InputFile_length_Datapart - ermittelt die min und max Länge vom Datenteil der eingelesenen RAWMSG´s

    InputFile_one_ClockPulse - sucht den angegebenen ClockPulse mit 15% Tolleranz aus der Input_Datei und filtert die RAWMSG in die Export_Datei

    InputFile_one_SyncPulse - sucht den angegebenen SyncPulse mit 15% Tolleranz aus der Input_Datei und filtert die RAWMSG in die Export_Datei

    TimingsList - erstellt eine Liste der Protokolldatei <signalduino_protocols.hash> im CSV-Format welche zum Import genutzt werden kann

    change_bin_to_hex - wandelt die binäre Eingabe in hexadezimal um

    change_dec_to_hex - wandelt die dezimale Eingabe in hexadezimal um

    change_hex_to_bin - wandelt die hexadezimale Eingabe in binär um

    change_hex_to_dec - wandelt die hexadezimale Eingabe in dezimal um

    invert_bitMsg - invertiert die eingegebene binäre Nachricht

    invert_hexMsg - invertiert die eingegebene hexadezimale Nachricht

    reverse_Input - kehrt die Eingabe um
     → Beispiel: aus 1234567 wird 7654321

**Info menu (Links zum anklicken)**

    Display doc SD_ProtocolData.pm - zeigt alle ausgelesenen Informationen aus der SD_ProtocolData.pm Datei an mit der Option, diese zu Dispatchen

    Display Information all Protocols - zeigt eine Gesamtübersicht der Protokolle an

    Display readed SD_ProtocolList.json - zeigt alle ausgelesenen Informationen aus SD_ProtocolList.json Datei an mit der Option, diese zu Dispatchen

    Check it - nach einem erfolgreichen Dispatch erscheint dieser Punkt um die Sensordaten mit den JSON Informationen zu vergleichen

closed 2019-11-04 by sidey79