RFD-FHEM/SIGNALduino_TOOL #10: X10 number of repeats to low

sidey79, 2019-06-13:
Die Anzahl der repeats ist hier mit eins zu gering angegeben, da sich zwei vollständige Signale in den Daten befinden. Der Wert 2 wäre passender.

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/7bc56ccb9320ebd6f3edff59da7da4097471e50b/FHEM/lib/SD_Device_ProtocolList.json#L327-L333

HomeAutoUser, 2019-11-27:
Wie ist hier der aktuelle Stand oder ist dies erledigt?

Prinzipiell sehe ich die erste Übertragung nicht als Wiederholung.

HomeAutoUser, 2020-12-29:
aktuelle Version ist stimmend

```
2020.12.29 10:02:26.985 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 39 -> X10 Protocol matches, trying to demodulate
2020.12.29 10:02:27.004 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 39 dmsg 29454A847BD700 length 56 dispatch(1/4) 
2020.12.29 10:02:27.200 3: RFXX10REC: RFXX10REC Unknown device RFXX10REC_d745, please define it
2020.12.29 10:02:27.206 2: autocreate: define RFXX10REC_d745 RFXX10REC ds10a d745 Window
2020.12.29 10:02:27.209 2: autocreate: define FileLog_RFXX10REC_d745 FileLog ./log/RFXX10REC_d745-%Y-%m.log RFXX10REC_d745
2020.12.29 10:02:27.236 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 39 dmsg 29454A847BD700 length 56 dispatch(2/4) 
2020.12.29 10:02:27.244 4: sduino_dummy: Dispatch, 29454A847BD700, Dropped due to short time or equal msg
2020.12.29 10:02:27.264 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 39 dmsg 27454A847BD6 length 48 dispatch(3/4) 
```

message_dispatch_repeats | 2 | 2020-12-29 10:02:27
-- | -- | --
message_dispatched | 3 | 2020-12-29 10:02:27

`      "dmsg":"29454A847BD700", "dispatch_repeats":"2", "user":"wherzig",`

closed 2020-12-29 by HomeAutoUser