RFD-FHEM/SIGNALduino_TOOL #11: Number of repeats doesn't match default behavior

sidey79, 2019-06-13:
Der erwartete Wert von 5 Wiederholungen, passt nicht zu den Standardeinstellung des SIGNALduino Modules.

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/7bc56ccb9320ebd6f3edff59da7da4097471e50b/FHEM/lib/SD_Device_ProtocolList.json#L367

In der Commandref ist vier als default für maxMuMsgRepeat angegeben.
Das ist eine Grundsatzfrage die wir klären müssen, wie das Testszenario aussieht.

a) Testen gegen defaultwerte
b) Testen gegen spezifische Wert, welche auch immer.

Ralf9, 2019-06-13:
Um alle enthaltenen Wiederholungen auszuwerten ist es sinnvoll beim dummysduino die maxMuMsgRepeat auf einen ausreichenden Wert zu erhöhen, z.B. 100

Das mit den repeats ist eine definitionssache.
Ich habe nur die gleichen Wiederholungen gezählt.
Ich habe das Wort Wiederholung wörtlich genommen.
Bei 4 enthaltenen msg ist die erste keine Wiederholung, z.B.

```
0123434353534353434343434343435343534343534353534353
6123434353534353434343434343435343534343534353534353 1.Wiederholung
6123434353534353434343434343435343534343534353534353 2.Wiederholung
6123434353534353434343434343435343534343534353534353 3.Wiederholung
612343435353435343434343434343534353434353435
```
hier habe ich mich verzählt, die erste Nachricht mit der 0 am Anfang ist auch gültig, also 3 repeats
https://github.com/RFD-FHEM/SIGNALduino_TOOL/issues/9

sidey79, 2019-06-14:
Ich kann deine Herleitung nachvollziehen.

Wie wir die Wiederholungen zählen sollten wir festlegen.
Ich beziehe mich im weiteren erst einmal auf den Standardwert von maxMuMsgRepeat.

Annahme:
Wenn die 1. Nachricht nicht als Wiederholung gezählt wird, dann ist vermutlich bei einigen Datensätzen der Sollwert falsch. Z.b. wenn repeat auf 1 steht, müssten zwei vollständige Nachrichten enthalten sein.

Die Rückgabe der dispatch sub sollte dann 2 betragen und maxMuMsgRepeat sollte von 4 auf 3 reduziert worden sein, da die erste Nachricht nicht als Wiederholung gezählt wird.

Ein maxMuMsgRepeat von 4 würde es dann auch erlauben, dass die dispatch Funktion 5x aufgerufen werden darf.

Aktuell ist das nicht der Fall.
maxMuMsgRepeat wird bereits bei der ersten Nachricht reduziert und die sub dispatch kann maximal 4 Mal aufgerufen werden.

Ralf9, 2019-06-14:
Dann sollten wir erstmal festlegen wie wir repeats definieren. 
Anzahl der Nachrichten oder
Anzahl der Wiederholungen.

HomeAutoUser, 2019-06-14:
> Dann sollten wir erstmal festlegen wie wir repeats definieren.

= Anzahl der Wiederholungen

Bei einer Nachricht mit 4x dem selben Inhalt haben wir eine "Stammnachricht" und 3 Wiederholungen.

sidey79, 2019-06-14:
Ich kann die Sicht was eine Wiederholung ist gut nachvollziehen.
Will diese auch nicht wiederlegen. Mathematisch ist das auch sicher sehr korrekt.

Ich würde aber gerne noch mal aus einem anderen Standpunkt das ganze beleuchten.

Wenn ich ein Satz vier mal schreibe und vier Sätze vor mir sehe, dann ist der 1. Satz für mich die 1. Wiederholung von vieren.
Wieso definiere ich den 1. Satz als Wiederholung? Für mich ist das so, weil ich bereits von Anfang an vor hatte, vier Sätze zu schreiben.. Ich wollte ja nicht einen Satz schreiben und dann drei mal wiederholen.

Naja, wie sollen wir das nun machen?
Wenn wir festlegen, dass die 1. Übertragung keine Wiederholung ist, müssten wir auch ein paar Logs anpassen.
`SIGNALduino_Log3 $name, 4, "$name: Decoded matched MU Protocol id $id dmsg $dmsg length $bit_length dispatch($nrDispatch/". AttrVal($name,'maxMuMsgRepeat', 4) . ")$rssiStr";
`

Bezüglich der Anzahl der Wiederholungen. Da würde ich gegen die Standardeinstellung vom Modul testen. Die haben wir ja auch aus gutem Grund so gewählt und ein Anwender sollte diese nicht ändern müssen,.

elektron-bbs, 2019-06-14:
Wofür soll es eigentlich gut sein, die Anzahl der Wiederholungen zu prüfen?
Durch die Begrenzung auf max. 4 ist das Original sowieso nicht abgebildet. Wir hatten heute gerade wieder einen Sensor auf dem Tisch, der die Nachricht 15 mal wiederholt, oder letztens der Klingeltaster mit >100...
Wichtig ist doch nur, was hinten raus kommt :-)
Das ist zum 1. die DMSG und dann die Werte, die das Modul zurück gibt (Temperatur, Feuchte etc.).

sidey79, 2019-06-14:
Das prüfen ist gut, um zu verifizieren ob weiterhin die gleiche Anzahl an Wiederholungen erkannt werden.

Kann ja durchaus sein, dass wir etwas verändern und dann keine Wiederholungen mehr erkannt werden oder nur noch die 1. etc.