RFD-FHEM/SIGNALduino_TOOL #12: Number of repeats opus XT300 to low

sidey79, 2019-06-13:
Hier sind zwei Wiederholungen im Signal die beide die gleiche DMSG ergeben:

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/7bc56ccb9320ebd6f3edff59da7da4097471e50b/FHEM/lib/SD_Device_ProtocolList.json#L392

HomeAutoUser, 2019-06-13:
Habe ich überprüft und angepasst.
Stimmt.

sidey79, 2019-06-14:
Unter der Prämisse, dass die 1. Übertragung nicht als Wiederholung gerechnet wird, müsste der Wert repeat wieder auf 1 gesetzt werden

HomeAutoUser, 2019-11-27:
Wie ist hier der aktuelle Stand oder ist dies erledigt?

Prinzipiell sehe ich die erste Übertragung nicht als Wiederholung.

sidey79, 2019-11-27:
So 100% ist das leider noch nicht gelöst.

Mittlerweile wurde der Eintrag in dispatch_repeats umbenannt.

Allerdings kann man effektiv die Anzahl an dispatch Aufrufen vergleichen, nicht wieviel davon wiederholungen sind.

HomeAutoUser, 2020-12-29:
Ich denke der Fall sollte erledigt sein.
Die Benennung wurde in 
- `dispatch_repeats` benannt, was die Anzahl an Wiederholung(en) an dispatch Aufrufen entspricht

closed 2020-12-29 by HomeAutoUser