RFD-FHEM/SIGNALduino_TOOL #9: FA22RF to less number of repats expected

sidey79, 2019-06-13:
Der Wert Repat besagt zwei, aber in der RMSG finden sich vier Wiederholungen die auch ausgewertet werden:

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/7bc56ccb9320ebd6f3edff59da7da4097471e50b/FHEM/lib/SD_Device_ProtocolList.json#L184-L191

` "repeat":"4"`  wäre der korrekte Wert

HomeAutoUser, 2019-06-13:
Habe ich überprüft und angepasst.
Stimmt.

sidey79, 2019-06-14:
Ok, also ich nehme an, Du  hast jetzt den Wert 3 verwendet mit der Sichtweise

1x Stammnachricht
3x Wiedeholung

Dann müsste ich `$AnzahlDispatches == $repeat+1 ` prüfen.

HomeAutoUser, 2019-11-27:
Wie ist hier der aktuelle Stand oder ist dies erledigt?

Prinzipiell sehe ich die erste Übertragung nicht als Wiederholung.

HomeAutoUser, 2020-12-29:
aktuelle Version

message_dispatch_repeats | 3 | 2020-12-29 10:09:02
-- | -- | --
message_dispatched | 4 | 2020-12-29 10:09:02
 
```
2020.12.29 10:09:02.564 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 13.1 -> FLAMINGO FA22RF / FA21RF / LM-101LD matches, trying to demodulate
2020.12.29 10:09:02.579 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 13.1 dmsg P13.1#CBFAD2 length 24 dispatch(1/4) RSSI = -74
2020.12.29 10:09:02.593 1: FLAMINGO UNDEFINED sensor detected, code CBFAD2, protocol 13.1
2020.12.29 10:09:02.599 2: autocreate: define FLAMINGO_CBFAD2 FLAMINGO CBFAD2
2020.12.29 10:09:02.600 2: autocreate: define FileLog_FLAMINGO_CBFAD2 FileLog ./log/FLAMINGO_CBFAD2-%Y-%m.log FLAMINGO_CBFAD2
2020.12.29 10:09:02.624 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 13.1 dmsg P13.1#CBFAD2 length 24 dispatch(2/4) RSSI = -74
2020.12.29 10:09:02.632 4: sduino_dummy: Dispatch, P13.1#CBFAD2, Dropped due to short time or equal msg
2020.12.29 10:09:02.647 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 13.1 dmsg P13.1#CBFAD2 length 24 dispatch(3/4) RSSI = -74
2020.12.29 10:09:02.655 4: sduino_dummy: Dispatch, P13.1#CBFAD2, Dropped due to short time or equal msg
2020.12.29 10:09:02.671 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 13.1 dmsg P13.1#CBFAD2 length 24 dispatch(4/4) RSSI = -74
2020.12.29 10:09:02.679 4: sduino_dummy: Dispatch, P13.1#CBFAD2, Dropped due to short time or equal msg
```

`      "dmsg":"P13.1#CBFAD2", "dispatch_repeats":"3", "user":"HomeAuto_User",`

closed 2020-12-29 by HomeAutoUser