RFD-FHEM/SIGNALduino_TOOL #14: TFA 30.3208.0 - number of repeats is not number of dispatches

sidey79, 2019-06-14:
Hier haben wir auch einen interessanten Fall:

Im Signal selbst ist in der Tat drei mal die gleiche Nachricht enthalten.
 
https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/b4314bddce9187765e2288cb85ad03c38a4a399c/FHEM/lib/SD_Device_ProtocolList.json#L408-L416

Allerdings verlässt diese Information (Anzahl Übertragungen) nicht die sub `SIGNALduino_MCTFA`

Es gibt ein paar Logmeldungen dazu, das war es aber auch.

```
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: TFA 30.3208.0 preamble_pos = 12
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: TFA 30.3208.0 preamble_pos = 12
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: TFA message start(1)=12 end=64 with length=52
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: TFA message start(1)=12 end=64 with length=52
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 5: dummyDuino: TFA message part(1)=0100010111001000000101000010010001000101110110110000
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 5: dummyDuino: TFA message part(1)=0100010111001000000101000010010001000101110110110000
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: TFA message part(1)=45C8142445DB0
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: TFA message part(1)=45C8142445DB0
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: TFA message start(2)=77 end=129 with length=52
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: TFA message start(2)=77 end=129 with length=52
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 5: dummyDuino: TFA message part(2)=0100010111001000000101000010010001000101110110110000
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 5: dummyDuino: TFA message part(2)=0100010111001000000101000010010001000101110110110000
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: TFA message part(2)=45C8142445DB0
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: TFA message part(2)=45C8142445DB0
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: TFA message start(3)=142 end=194 with length=52
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: TFA message start(3)=142 end=194 with length=52
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 5: dummyDuino: TFA message part(3)=0100010111001000000101000010010001000101110110110000
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 5: dummyDuino: TFA message part(3)=0100010111001000000101000010010001000101110110110000
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: TFA message part(3)=45C8142445DB0
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: TFA message part(3)=45C8142445DB0
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 4: dummyDuino: repeated hex 45C8142445DB0 found 3 times
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 4: dummyDuino: repeated hex 45C8142445DB0 found 3 times
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 5: dummyDuino Dispatch: W58#45C8142445DB0, test ungleich: disabled
2019.06.14 23:57:23 5: End notify loop for dummyDuino
2019.06.14 23:57:23 5: dummyDuino Dispatch: W58#45C8142445DB0, test ungleich: disabled
2019.06.14 23:57:23 5: Starting notify loop for dummyDuino, 1 event(s), first is dummyDuino 5: dummyDuino Dispatch: W58#45C8142445DB0, -57 dB, dispatch
2019.06.14 23:57:23 5: End notify loop for dummyDuino
```

Effektiv gibt es einen einzigen dispatch Aufruf:

```
2019.06.14 23:57:23 5: dummyDuino Dispatch: W58#45C8142445DB0, -57 dB, dispatch
2019.06.14 23:57:23 5: dummyDuino: dispatch W58#45C8142445DB0
```

Der Wert lässt sich nicht über die Anzahl der Dispatches prüfen. Können wir den Wert auf 1 setzen?

HomeAutoUser, 2019-06-19:
Ich habe nun diese Nachricht auch mal dispatcht und komme auf den Wert 1.
Wie dort auf den Wert 2 gekommen wird, da gibt es nur die Möglichkeiten
- verschrieben
- andere Anpassung in der Auswertung
- ...

HomeAutoUser, 2021-03-25:
```
2021.03.25 14:20:16.261 4: sduino_dummy: get rawmsg: MC;LL=-981;LH=964;SL=-480;SH=520;D=002BA37EBDBBA24F0015D1BF5EDDD127800AE8DFAF6EE893C;C=486;L=194;R=34;
2021.03.25 14:20:16.266 4: sduino_dummy: Parse_MC, Found manchester protocol id 10 clock 486 RSSI = -57 -> Oregon Scientific v2|v3
2021.03.25 14:20:16.277 4: sduino_dummy: Parse_MC, Found manchester protocol id 12 clock 486 RSSI = -57 -> Hideki
2021.03.25 14:20:16.293 4: sduino_dummy: Parse_MC, Found manchester protocol id 52 clock 486 RSSI = -57 -> Oregon Scientific PIR
2021.03.25 14:20:16.305 4: sduino_dummy: Parse_MC, Found manchester protocol id 58 clock 486 RSSI = -57 -> TFA 30.3208.0
2021.03.25 14:20:16.312 4: sduino_dummy: lib/mcBit2TFA, 30.3208.0 preamble_pos = 12
2021.03.25 14:20:16.316 4: sduino_dummy: lib/mcBit2TFA, message start(1)=12 end=64 with length=52
2021.03.25 14:20:16.324 4: sduino_dummy: lib/mcBit2TFA, message part(1)=45C8142445DB0
2021.03.25 14:20:16.328 4: sduino_dummy: lib/mcBit2TFA, message start(2)=77 end=129 with length=52
2021.03.25 14:20:16.336 4: sduino_dummy: lib/mcBit2TFA, message part(2)=45C8142445DB0
2021.03.25 14:20:16.340 4: sduino_dummy: lib/mcBit2TFA, message start(3)=142 end=194 with length=52
2021.03.25 14:20:16.347 4: sduino_dummy: lib/mcBit2TFA, message part(3)=45C8142445DB0
2021.03.25 14:20:16.351 4: sduino_dummy: lib/mcBit2TFA, repeated hex 45C8142445DB0 found 3 times"
2021.03.25 14:20:16.543 4: sduino_dummy: SD_WS_Parse protocol 58, rawData 45C8142445DB0
```

closed 2021-03-25 by HomeAutoUser