RFD-FHEM/SIGNALduino_TOOL #15: RH787T number of repeats is not number of dispatches

sidey79, 2019-06-14:
https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/b4314bddce9187765e2288cb85ad03c38a4a399c/FHEM/lib/SD_Device_ProtocolList.json#L592-L600

Hier haben wir auch ein Problem.
Die Daten führen dazu, dass 4x ein Dispatch
`dummyDuino: dispatch P30#077 `und 4x  `dispatching hex: P83#077
`
erfolgt.
Das macht dann in Summe 8 dispatches und passt nicht zu repeats.

HomeAutoUser, 2019-06-19:
Ich komme bei dem Signal auf 7 Wiederholungen bei 8x dispatch.

Ralf9, 2019-06-20:
In dem Signal sind 7 Wiederholungen enthalten:
```
01230353535353562626235626262
72 353535353562626235626262
72 353535353562626235626262 1. repeat
72 353535353562626235626262 2. repeat
72 353535353562626235626262 3. repeat
72 353535353562626235626262 4. repeat
72 353535353562626235626262 5. repeat
72 353535353562626235626262 6. repeat
72 353535353562626235626262 7. repeat
72 35353535356262;CP=2;R=43;
```

@sidey79 
Verwendest Du für Deine Tests eine Whitelist?
Ohne Whitelist macht es für mich keinen Sinn, wenn eine neue Protokoll ID mit einem ähnlichen Protokoll dazukommt, könnte sich die Anzahl der repeats erhöhen.

HomeAutoUser, 2021-03-25:
```
2021.03.25 14:30:27.517 4: sduino_dummy: get rawmsg: MU;P0=388;P1=-112;P2=267;P3=-378;P5=585;P6=-693;P7=-11234;D=0123035353535356262623562626272353535353562626235626262723535353535626262356262627235353535356262623562626272353535353562626235626262723535353535626262356262627235353535356262623562626272353535353562626235626262723535353535626262356262627235353535356262;CP=2;R=43;O;
2021.03.25 14:30:27.525 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 8 -> TX3 Protocol matches, trying to demodulate
2021.03.25 14:30:27.655 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 40 -> Romotec  matches, trying to demodulate
2021.03.25 14:30:27.723 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 54 -> TFA 30.3233.01 matches, trying to demodulate
2021.03.25 14:30:27.752 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 61 -> FS10 matches, trying to demodulate
2021.03.25 14:30:27.788 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 70 -> FHT80TF matches, trying to demodulate
2021.03.25 14:30:27.810 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 73 -> FHT80 matches, trying to demodulate
2021.03.25 14:30:27.824 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 74 -> FS20 matches, trying to demodulate
2021.03.25 14:30:27.860 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 80 -> EM1000WZ matches, trying to demodulate
2021.03.25 14:30:27.882 4: sduino_dummy: Parse_MU, Fingerprint for MU protocol id 83 -> RH787T matches, trying to demodulate
2021.03.25 14:30:27.910 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(1/10) RSSI = -52.5
2021.03.25 14:30:27.924 4: sduino_dummy: Dispatch, P83#077, Dropped due to short time or equal msg
2021.03.25 14:30:27.952 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(2/10) RSSI = -52.5
2021.03.25 14:30:27.968 4: sduino_dummy: Dispatch, P83#077, Dropped due to short time or equal msg
2021.03.25 14:30:27.997 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(3/10) RSSI = -52.5
2021.03.25 14:30:28.011 4: sduino_dummy: Dispatch, P83#077, Dropped due to short time or equal msg
2021.03.25 14:30:28.040 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(4/10) RSSI = -52.5
2021.03.25 14:30:28.054 4: sduino_dummy: SD_UT protocol 83, bitData 000001110111, hlen 3
2021.03.25 14:30:28.087 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(5/10) RSSI = -52.5
2021.03.25 14:30:28.101 4: sduino_dummy: Dispatch, P83#077, Dropped due to short time or equal msg
2021.03.25 14:30:28.130 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(6/10) RSSI = -52.5
2021.03.25 14:30:28.144 4: sduino_dummy: Dispatch, P83#077, Dropped due to short time or equal msg
2021.03.25 14:30:28.172 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(7/10) RSSI = -52.5
2021.03.25 14:30:28.186 4: sduino_dummy: Dispatch, P83#077, Dropped due to short time or equal msg
2021.03.25 14:30:28.214 4: sduino_dummy: Parse_MU, Decoded matched MU protocol id 83 dmsg P83#077 length 12 dispatch(8/10) RSSI = -52.5
2021.03.25 14:30:28.229 4: sduino_dummy: Dispatch, P83#077, Dropped due to short time or equal msg
```

closed 2021-03-25 by HomeAutoUser