RFD-FHEM/SIGNALduino_TOOL #21: Automatisches Tests der JSON Datei

sidey79, 2019-11-04:
Von #20:

Es soll automatisiert geprüft werden, ob die JSON Datei dem korrekten Syntax entspricht.

- Der Test soll in Perl laufen
- Der Test soll nach jedem commit auf travis-ci laufen

sidey79, 2019-11-04:
@HomeAutoUser 

Schau mal ob das deinen Vorstellungen so entspricht:

https://travis-ci.com/RFD-FHEM/SIGNALduino_TOOL/builds/134932501
https://github.com/RFD-FHEM/SIGNALduino_TOOL/runs/288281510
https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/b7522aa869ce0d57b71faa267198137a89ecd95b/t/json_check.pl

HomeAutoUser, 2019-11-05:
Ich denke, das sollte für den Anfang reichen.
Abändern können wir ja immer nochmal.

HomeAutoUser, 2019-11-05:
Achso, ne kleine Anmerkung habe ich noch.
Können wir https://github.com/RFD-FHEM/SIGNALduino_TOOL/tree/pre-release/t den Ordner t einen "festen Namen" verpassen so wie bei den anderen Repros. Bsp: Tests oder UnitTest
Ich fände es von vorteil, wenn dieser bei unseren Tests gleich ist um sofort zu erkennen AH... das ist das ;-)

sidey79, 2019-11-05:
Ich habe mich da ein wenig weiter entwickelt und mich an vorhandene Standards gehalten.
Der Ordner t ist üblich um Tests abzulegen.
in t/lib kommen besondere Bibliotheken.

https://github.com/Perl/perl5/blob/blead/t/README

Würde es so lassen, da es ein 100% Perl Test ist und nicht wie beim UnitTest Modul ein Test innerhalb von FHEM.

HomeAutoUser, 2019-11-05:
Dnn belassen wir es so und wieder wurden wir weitergebildet :-)

HomeAutoUser, 2019-11-30:
@sidey79 laufen hier im Hintergrund schon Tests wenn neue PR´s mit RAWMSG´s für die JSON Datei  durchgeführt werden?

sidey79, 2019-11-30:
Im Branch pre-release laufen die.

Nachdem ich mich die Tage ein wenig mit GitHub Actions beschäftigt habe, finde ich, dass wir das umstellen sollten. :)

HomeAutoUser, 2019-11-30:
"Hier würde" ich mir ja das umstellen auf GitAction ja selbst zutrauen aber ich weiß nicht, was die bisherigen angelegten Tests dann machen?

Den Vorteil in GitAction sehe ich auf jedenfall, das die controls_Datei angepasst wird und zusätzlich auch die CHANGED ;-)

sidey79, 2019-11-30:
Naja, die Controls Datei wird aktuell ja mittels travisCI aktualisiert, aber das überschreibt das letzte Ergebnis vom Check. :(

Die Anpassungen für Tests habe ich im RSL Repo ein wenig erprobt und kann es als Blaupause verwenden.

HomeAutoUser, 2019-12-13:
Thema "Automatisches Tests der JSON Datei", diese laufen im Hintergrund. https://github.com/RFD-FHEM/SIGNALduino_TOOL/issues/21#issuecomment-559991850

Somit schließe ich das Thema. Was die automatische Anpassung von Dateien wie Controls oder Changed angeht, so können wir dies in einem neuen Faden erörtern sobald wir eine Lösung haben.

closed 2019-12-13 by HomeAutoUser