RFD-FHEM/SIGNALduino_TOOL #35: Button "Check it" failed

HomeAutoUser, 2019-12-17:
@elektron-bbs,

bitte mal verifizieren.
Ich suche mir gerade den "Wolf" und bin der Meinung, wir hatten das als letzte Tat von uns zusammen nochmal geändert.

HomeAutoUser, 2019-12-17:
Bitte nicht weiter suchen, Fehler gefunden und ich schiebe den fix gleich hoch.

elektron-bbs, 2019-12-17:
Jetzt, wo ich gerade eine Vermutung hatte :-)
Liegt es an der Umstellng der Logausgaben?

HomeAutoUser, 2019-12-17:
Ja genau daran lag es.
Zugleich habe ich gleich noch etwas Notizen vorgenommen damit es auch deutlich bleibt ;-)

elektron-bbs, 2019-12-17:
Das passt zwar nicht hier dazu, aber ich habe auch gerade ein Problem:
Meine zwei noch auf Debian 8 basierenden Geräte machen kein Update von FHEM mehr. Ich bekomme immer folgende Fehlermeldung:

https://fhem.de/fhemupdate/controls_fhem.txt: Can't connect(2) to https://fhem.de:443:  SSL connect attempt failed error:14077410:SSL routines:SSL23_GET_SERVER_HELLO:sslv3 alert handshake failure

Vieleicht hast du ja auf Anhieb eine Idee?

HomeAutoUser, 2019-12-17:
Eine direkte Idee habe ich nicht soweit es das komplette FHEM ist.
Soben wurde nur das DEV-r34 gefixt weil da FHEM meldete falsche Größe der controls Datei.

Vielleicht wirst du mit dem Suchbefriff `SSL connect attempt failed error:14077410:SSL` fündig. Da gibt es einige Treffe und beim überfliegen scheint es ggf. PERL zu sein oder ein Debianpaket.

sidey79, 2019-12-17:
Du musst eine neuere SSL bzw. TLS Version verwenden. z.B. v1.2 
Eventuell fehlen dir da auch Bibliotheken.

closed 2019-12-17 by HomeAutoUser