RFD-FHEM/SIGNALduino_TOOL #37: Can't locate lib/SD_Protocols.pm ...

plin2, 2019-12-21:
Habe gerade versucht das SIGNALduino_TOOL in meiner FHEM-Instanz zu installieren. Beim **reload 88_SIGNALduino_TOOL** bekomme ich folgende Fehlermeldung:
Can't locate lib/SD_Protocols.pm in @INC (you may need to install the lib::SD_Protocols module) (@INC contains: ./FHEM/lib ./lib . /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base ./FHEM) at ./FHEM/88_SIGNALduino_TOOL.pm line 27.
BEGIN failed--compilation aborted at ./FHEM/88_SIGNALduino_TOOL.pm line 27.
Im per git clone kopierten Ordner FHEM/lib gibt es nur ein SD_Device_ProtocolList.json

HomeAutoUser, 2019-12-21:
Guten Morgen @plin2,
kannst du bitte mal ein list von deinem SIGNALduino machen?
Mfg

plin2, 2019-12-21:
ok, das war's. Welche Version der SIGNALduino firmware soll ich mir runterladen?

Das Listing des Devices sieht so aus

Internals:
   Clients    :IT:CUL_TCM97001:SD_RSL:OREGON:CUL_TX:SD_AS:Hideki:SD_WS07:SD_WS09: :SD_WS:RFXX10REC:Dooya:SOMFY:SD_BELL:SD_UT:SD_WS_Maverick:FLAMINGO:CUL_WS:Revolt: :FS10:CUL_FHTTK:Siro:FHT:FS20:CUL_EM:Fernotron:SD_Keeloq:SD_GT:SIGNALduino_un:
   DEF        /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0@57600
   DMSG       nothing
   DevState   initialized
   DeviceName /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0@57600
   FD         11
   FUUID      5c66e6dc-f33f-cfec-c9fb-8a0795a919d314a0
   IDsNoDispatch 2,72.1,82
   LASTDMSG   nothing
   LASTDMSGID nothing
   NAME       mySIGNALduino
   NR         38
   PARTIAL    
   STATE      opened
   TIME       1576939512
   TYPE       SIGNALduino
   hasCC1101  1
   sendworking 0
   version    V 3.3.1-RC-nightly SIGNALduino cc1101 (chip CC1101) - compiled at Feb  3 2019 21:35:11
   versionProtocols 1.12
   versionmodul v3.4.1_dev_18.12
   MatchList:
     10:SD_WS07 ^P7#[A-Fa-f0-9]{6}[AFaf][A-Fa-f0-9]{2,3}
     11:SD_WS09 ^P9#F[A-Fa-f0-9]+
     12:SD_WS   ^W\d+x{0,1}#.*
     13:RFXX10REC ^(20|29)[A-Fa-f0-9]+
     14:Dooya   ^P16#[A-Fa-f0-9]+
     15:SOMFY   ^Ys[0-9A-F]+
     16:SD_WS_Maverick ^P47#[A-Fa-f0-9]+
     17:SD_UT   ^P(?:14|20|29|30|34|46|68|69|76|81|83|86|90|91|91.1|92|93|95)#.*
     18:FLAMINGO ^P13\.?1?#[A-Fa-f0-9]+
     19:CUL_WS  ^K[A-Fa-f0-9]{5,}
     1:IT       ^i......
     20:Revolt  ^r[A-Fa-f0-9]{22}
     21:FS10    ^P61#[A-F0-9]+
     22:Siro    ^P72#[A-Fa-f0-9]+
     23:FHT     ^81..(04|09|0d)..(0909a001|83098301|c409c401)..
     24:FS20    ^81..(04|0c)..0101a001
     25:CUL_EM  ^E0.................
     26:Fernotron ^P82#.*
     27:SD_BELL ^P(?:15|32|41|42|57|79|96)#.*
     28:SD_Keeloq ^P(?:87|88)#.*
     29:SD_GT   ^P49#[A-Fa-f0-9]+
     2:CUL_TCM97001 ^s[A-Fa-f0-9]+
     3:SD_RSL   ^P1#[A-Fa-f0-9]{8}
     4:OREGON   ^(3[8-9A-F]|[4-6][0-9A-F]|7[0-8]).*
     5:CUL_TX   ^TX..........
     6:SD_AS    ^P2#[A-Fa-f0-9]{7,8}
     7:Hideki   ^P12#75[A-F0-9]+
     9:CUL_FHTTK ^T[A-F0-9]{8}
     X:SIGNALduino_un ^[u]\d+#.*
   QUEUE:
   READINGS:
     2019-03-31 09:18:04   ccconf          freq:868.300MHz bWidth:58KHz rAmpl:33dB sens:4dB  (DataRate:5603.79Baud)
     2019-03-28 20:42:48   ccpatable       C3E = 09 00 2F 00 00 00 00 00
     2018-02-08 11:38:24   ccreg           C3E = 00 C0 00 00 00 00 00 00
     2018-04-15 18:03:47   cmds            V R t X F S P C r p W x e
     2019-03-21 19:09:55   config          MS=0;MU=1;MC=0;Mred=1
     2018-02-05 21:16:37   freeram         293
     2019-12-21 15:46:25   ping            OK
     2019-03-21 19:01:25   raw             C20n11=FB5611EF2D121F4100597F3E88310B00FF
     2019-12-21 15:45:29   statStateDay    closed: 1d 04:25:13 closed_Count: 1 disconnected: 71d 13:04:54 disconnected_Count: 42 opened: 346d 03:27:05 opened_Count: 44 (since: 2018-10-28_18:48:17)
     2019-12-21 15:45:29   statStateMonth  closed: 1d 04:25:13 closed_Count: 1 disconnected: 71d 13:04:54 disconnected_Count: 42 opened: 346d 03:27:05 opened_Count: 44 (since: 2018-10-28_18:48:17)
     2019-12-21 15:45:29   statStateYear   closed: 1d 04:25:13 closed_Count: 1 disconnected: 71d 13:04:54 disconnected_Count: 42 opened: 346d 03:27:05 opened_Count: 44 (since: 2018-10-28_18:48:17)
     2019-12-21 15:45:25   state           opened
     2018-02-05 21:17:45   uptime          0 00:03:09
     2019-12-21 15:45:25   version         V 3.3.1-RC-nightly SIGNALduino cc1101 (chip CC1101) - compiled at Feb  3 2019 21:35:11
   getcmd:
   helper:
     _98_statistics myStatDevice
   keepalive:
     ok         1
     retry      0
   mcIdList:
     10
     11
     12
     18
     43
     47
     52
     57
     58
     96
   msIdList:
     0
     0.1
     0.2
     0.3
     0.4
     1
     3
     3.1
     4
     6
     7
     13
     13.2
     14
     15
     17
     20
     23
     25
     33
     33.1
     33.2
     35
     41
     49
     51
     53
     55
     65
     68
     74.1
     87
     88
     90
     91.1
     93
   muIdList:
     8
     9
     13.1
     16
     17.1
     19
     21
     22
     24
     28
     29
     30
     32
     34
     36
     37
     38
     39
     40
     42
     44
     44.1
     45
     46
     48
     49.1
     49.2
     50
     56
     59
     60
     61
     62
     64
     66
     67
     69
     70
     71
     72
     73
     74
     76
     79
     80
     81
     83
     84
     85
     86
     89
     91
     92
     94
     95
Attributes:
   WS09_CRCAUS 0
   flashCommand avrdude -c arduino -b [BAUDRATE] -P [PORT] -p atmega328p -vv -U flash:w:[HEXFILE] 2>[LOGFILE]
   hardware   nanoCC1101
   room       SIGNALduino,Test
   verbose    5

HomeAutoUser, 2019-12-21:
> ok, das war's.

fein :-)

> Welche Version der SIGNALduino firmware soll ich mir runterladen?

Die Firmware hat mit dem TOOL nichts direkt zu tun. Natürlich ist es ratsam, wenn man mitentwickeln oder testen möchte, öfter zeitnah die Entwicklerversionen zu verwenden.

Was ist deine Intension das SIGNALduino_TOOL zu nutzen?

plin2, 2019-12-21:
Unterstützung von 2FSK bzw. GFSK. Deshalb geht's im anderen Thread weiter. Muss jetzt erst mal schauen wie mir das SIGNALduino_TOOL  hilft.

closed 2019-12-21 by plin2