RFD-FHEM/SIGNALduino_TOOL #41: QUIGG GT-9000

HomeAutoUser, 2020-05-02:
@elektron-bbs 

kannst du bitte zu folgendem Protokoll
```

		"49"	=>	## QUIGG GT-9000, EASY HOME RCT DS1 CR-A, uniTEC 48110 and other
							# The remote sends 8 messages in 2 different formats.
							# SIGNALduino decodes 4 messages from remote control as MS then ...
							# https://github.com/RFD-FHEM/RFFHEM/issues/667 - Oct 19, 2019
							# DMSG: 5A98B0   MS;P0=-437;P3=-1194;P4=1056;P6=297;P7=-2319;D=67634063404063406340636340406363634063404063636363;CP=6;SP=7;R=37;
							# DMSG: 887F92   MS;P1=-2313;P2=1127;P3=-405;P4=379;P5=-1154;D=41234545452345454545232323232323232345452345452345;CP=4;SP=1;R=251;
							# DMSG: E6D12E   MS;P0=1062;P1=-1176;P2=315;P3=-2283;P4=-433;D=23040404212104042104042104212121042121042104040421;CP=2;SP=3;R=26;
			{
				name            => 'GT-9000',
				comment         => 'Remote control EASY HOME RCT DS1 CR-A',
				id              => '49',
				knownFreqs      => '433.92',
				clockabs        => 383,
				one             => [3,-1],  # 1150,-385 (timings from salae logic)
				zero            => [1,-3],  # 385,-1150 (timings from salae logic)
				sync            => [1,-6],  # 385,-2295 (timings from salae logic)
				format          => 'twostate',
				preamble        => 'P49#',
				clientmodule    => 'SD_GT',
				modulematch     => '^P49.*',
				length_min      => '24',
				length_max      => '24',
			},
```

mir 2 RAWMSG liefern welche dann zusammen in dem SD_GT Modul verarbeitet werden ohne das der State 
`version not unique, please press another button`
oder
`learned code 1, please press another button`
erscheint?

Die Daten im JSON 

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/770d865aab4ea9b92e7065b13cd172b8a42e43d2/FHEM/lib/SD_Device_ProtocolList.json#L726-L732

sind nicht aktuelle und unzureichend ;-)

Besser ist es vielleicht, das wir 4 RAWMSG besitzen wo deine V1 und V2 abgedeckt sind.

elektron-bbs, 2020-05-02:
Nachrichten kannst du dafür jede Menge bekommen. Ein automatischer Test dafür wird wahrscheinlich ziemlich aufwändig. Ich zitiere mal aus der Doku:

> Das Einrichten der Fernbedienung erfolgt in einem Lernprozess. Nach dem Empfang von mindestens 5 Nachrichten innerhalb von 3 Minuten wird ein neues Gerät "SD_GT_LEARN" angelegt. Das Einrichten der einzelnen Tasten der Fernbedienung beginnt nach dem Empfang weiterer 6 verschiedener Nachrichten. Dieser Lernprozess wird mit dem Status "learned code 4, please press another button" signalisiert, wobei der Zähler die Anzahl der aktuell registrierten Codes anzeigt. 
> Es müssen jetzt sämtliche Tasten der Fernbedienung mehrmals betätigt werden. Bei erfolgreicher Dekodierung der Funksignale werden dabei die einzelnen Tasten angelegt.

[penny_C_off.txt](https://github.com/RFD-FHEM/SIGNALduino_TOOL/files/4568584/penny_C_off.txt)
[penny_C_on.txt](https://github.com/RFD-FHEM/SIGNALduino_TOOL/files/4568586/penny_C_on.txt)
[penny_D_off.txt](https://github.com/RFD-FHEM/SIGNALduino_TOOL/files/4568587/penny_D_off.txt)
[penny_D_on.txt](https://github.com/RFD-FHEM/SIGNALduino_TOOL/files/4568589/penny_D_on.txt)
[Quigg_GT-9000_1.txt](https://github.com/RFD-FHEM/SIGNALduino_TOOL/files/4568591/Quigg_GT-9000_1.txt)
[OBI-Emil-Lux.txt](https://github.com/RFD-FHEM/SIGNALduino_TOOL/files/4568594/OBI-Emil-Lux.txt)

HomeAutoUser, 2020-05-02:
Vielen Dank,
mir geht es erstmal um die Doku und ein Test ... nachranging.

Frage:
Wieso passiert hier mit den Nachrichten nichts?
```
MS;P0=1059;P2=-434;P3=313;P4=-1180;P5=-2283;D=35020202343434343402343434340234343434340202023434;CP=3;SP=5;R=25;4;4;
MS;P1=1059;P2=-434;P3=319;P4=-1178;P5=-2283;D=35121212343412343434341234123434341234343412123434;CP=3;SP=5;R=23;m2;4;4;
MS;P0=317;P1=-2283;P2=1059;P3=-434;P4=-1176;D=01232323042304042323042304042323042323230404230423;CP=0;SP=1;R=22;m2;
MS;P2=1060;P4=314;P5=-2282;P6=-434;P7=-1175;D=45262626474747262626264747264747262647472647264726;CP=4;SP=5;R=26;G;
MS;P0=1062;P1=-1176;P2=315;P3=-2283;P4=-433;D=23040404212104042104042104212121042121042104040421;CP=2;SP=3;R=26;
MS;P0=-2156;P1=316;P3=1059;P4=-433;P5=-1177;D=10343434151534153415341534341534153415341534343415;CP=1;SP=0;R=32;m2;
```

Zitat Doku:
>Das Einrichten der Fernbedienung erfolgt in einem Lernprozess. Nach dem Empfang von mindestens 5 Nachrichten innerhalb von 3 Minuten wird ein neues Gerät "SD_GT_LEARN" angelegt.

closed 2020-12-29 by HomeAutoUser