RFD-FHEM/SIGNALduino_TOOL #42: Dispatch DMSG

elektron-bbs, 2020-05-10:
Der Dispatch von DMSG aus der Datei SD_Device_ProtocolList.json funktioniert bei mir nicht mehr 100%ig.

```
2020.05.10 17:19:59 3: sduino_dummy: Unknown code P46, help me!
2020.05.10 17:26:43 3: sduino_dummy: Unknown code P7, help me!
2020.05.10 17:28:11 3: sduino_dummy: Unknown code U49, help me!
2020.05.10 21:12:55 3: sduino_dummy: Unknown code P9, help me!
2020.05.10 21:14:00 3: sduino_dummy: Unknown code P14, help me!
2020.05.10 21:14:22 3: sduino_dummy: Unknown code W27, help me!
```

Es wird offensichtlich nur der Teil vor "#" übergeben. Dispatche ich Nachrichten direkt miitels

`set SIGNALduino_TOOL Dispatch_DMSG  P9#FF4A63E4AA040802F40168`

funktioniert es. DMSG für z.B. CUL_TCM97001, IT oder OREGON sind davon nicht betroffen.

Ich bin mir fast sicher, das das vor wenigen Tagen noch funktioniert hat. Probiert habe ich sowohl master als auch pre-release auf 2 Systemen. Keine Ahnung, welches Update dafür verantwortlich ist. Am SIGNALduuino_TOOL kann es eigentlich kaum liegen, daran wurde ja seit Monaten nichts geändert.

sidey79, 2020-05-10:
Welchen Branch betrifft es? Alle?

Ich habe am UnitTest modul eine kleinigkeit angepasst. Daran sollt es aber eher nicht liegen denke ich.
Wie genau hast Du getestet?

HomeAutoUser, 2020-05-10:
@elektron-bbs 
kannst du genau benennen was die Reihenfolge der Handlung ist um es nachzustellen?

Wie versteht sich:
`funktioniert bei mir nicht mehr 100%ig.` ?

Meinst du vielleicht, das du auf den Button `dmsg` klickst und nichts passiert?
Steht was im state oder nähere Informationen im Logfile?

PS: Bei einem Schnelltest mit der Vorab dev-r35... ging es soeben. Ich werde nochmal die dev-r34 testen.

elektron-bbs, 2020-05-11:
Also, ich lade die SD_Device_ProtocolList.json im SIGNALduino-Tool, klicke auf "Display readed SD_ProtocolList.json", Liste wird angezeigt, scrolle nach unten bis Id 7 und klicke dann auf den Button "dmsg".

Im Log habe ich dann folgenden Eintrag:

`2020.05.11 11:24:17 3: sduino_dummy: Unknown code P7, help me!
`

Ich habe dann probehalber mal in der SD_Device_ProtocolList.json bei dem zweiten Eintrag der Id 7 
von
`"dmsg":"P7#B180F4A38",
`
geändert in
`"dmsg":"P7B1#80F4A38",
`
Beim Dispatch dieser Nachricht erhalte ich im Log:

`2020.05.11 11:25:16 3: sduino_dummy: Unknown code P7B1, help me!
`

Fakt ist also, das mir irgend etwas die DMSG ab dem Zeichen "#" abhackt. Jetzt muss ich nur noch heraus bekommen, warum und wieso das passiert.
Ich habe das Verhalten übrigens heute auch noch auf unserem Werkstattrechner nachvollziehen können. Dort ist es genau so.

elektron-bbs, 2020-05-11:
Ich muss passen, ich hab jetzt den halben Tag verbracht, um den Fehler einzugrenzen, finde ihn aber nicht.
Im SIGNALduino_TOOL wird der Befehl mit dieser Zeit gebaut:

```
$buttons.= "<INPUT type=\"reset\" onclick=\"pushed_button(".@$ProtocolListRead[$i]->{id}.",'SD_Device_ProtocolList.json','dmsg','".@$ProtocolListRead[$i]->{name}."'); FW_cmd('/fhem?XHR=1&cmd.$name=set%20$name%20$NameDispatchSet"."DMSG%20$dmsg$FW_CSRF')\" value=\"dmsg\" %s/>" if ($dmsg ne "" && $Dummyname ne "none");`
```

der dann so aussehen sollte:

`set SIGNALduino_TOOL Dispatch_DMSG $dmsg$FW_CSRF`

Im fhemweb.js function FW_cmd kommt das auch noch richtig an:

[image]

01_FHEMWEB.pm Sub FW_answerCall($) wirft mir aber dann diesen Fehler ins Log:

```
2020.05.11 17:41:10 4: WEB_192.168.178.41_61613 POST /fhem?XHR=1&cmd.SIGNALduino_TOOL=set%20SIGNALduino_TOOL%20Dispatch_DMSG%20P7; BUFLEN:0
2020.05.11 17:41:10 3: FHEMWEB WEB CSRF error:  ne csrf_103440076041825 for client WEB_192.168.178.41_61613 / command set SIGNALduino_TOOL Dispatch_DMSG P7. For details see the csrfToken FHEMWEB attribute.
2020.05.11 17:41:10 4: WEB: /fhem?XHR=1&cmd.SIGNALduino_TOOL=set%20SIGNALduino_TOOL%20Dispatch_DMSG%20P7 / RL:20 / text/html; charset=UTF-8 / Content-Encoding: gzip
```

Es fehlt dann also praktisch von den ursprünglichen 2 Variablen $dmsg$FW_CSRF alles ab der Raute hinter P7.... in der DMSG.

HomeAutoUser, 2020-05-11:
@elektron-bbs, hat dein Versuch ein altes Backup einspielen Erkenntnisse über den Zeilraum bringen können?

sidey79, 2020-05-11:
Also hat das denn jemals funktioniert?

Meine 5 ct.

'#' Kann in einer Url als Zeichen nicht verwendet werden. 

Die ganze URL wird schon teilweise mit encodierten Zeichen aufgebaut z.B. %20.
Daraus schließe ich, dass hier am urlencode gespart wurde.

FHEM bietet sogar eine nette Funktion dafür an.
https://wiki.fhem.de/wiki/HttpUtils#urlEncode

Einfach ausprobieren kann man es im übrigen hier, wie z.B. eine Raute encoded wird:
https://www.urlencoder.org/

`
my $enc_url=::urlencode(qq[/fhem?XHR=1&cmd.$name=set $name ${$NameDispatchSet}DMSG $dmsg$FW_CSRF]);

$buttons.= qq[<INPUT type="reset" onclick="pushed_button(@$ProtocolListRead[$i]->{id}",'SD_Device_ProtocolList.json','dmsg','@$ProtocolListRead[$i]->{name}'); FW_cmd($enc_url)\" value="dmsg" %s/>] if ($dmsg ne '' && $Dummyname ne 'none');`
`

So in etwa, wobei es sein kann, dass ich leerzeichen oder so jetzt nicht richtig interpretiert habe  beim entwuseln von der Zeichenkette mit " ' und . ;-)

HomeAutoUser, 2020-05-11:
Ob alles funktioniert hat wird schwer zu beantworten sein. Vermutlich hat mal immer Dir funktionierenden Protokolle durch Zufall genommen.

Mit dem encoden # ist ein guter Hinweis. Ich kann man versuchen, mir dies in Ruhe anzusehen. Vielleicht wurde ja auch die Funktion in der FHEM_Web bzw. Passenden js Datei verändert...

elektron-bbs, 2020-05-12:
Die Zeile "$buttons.=........." ist schon ein arges Gewurschtel :-)
Diese durch @sidey79 Zeilen zu ersetzen ist mir auf Anhieb nicht gelungen. Ich habe es jetzt erst mal so gelöst, indem ich nur die DMSG durch urlEncode jage:
```
$buttons.= "<INPUT type=\"reset\" onclick=\"pushed_button(".@$ProtocolListRead[$i]->{id}.",'SD_Device_ProtocolList.json','dmsg','".@$ProtocolListRead[$i]->{name}."'); FW_cmd('/fhem?XHR=1&cmd.$name=set%20$name%20$NameDispatchSet"."DMSG%20".urlEncode($dmsg)."$FW_CSRF')\" value=\"dmsg\" %s/>" if ($dmsg ne "" && $Dummyname ne "none");
```

Wenn auch nicht schön, aber es funktioniert erst mal wieder.

HomeAutoUser, 2020-05-13:
@elektron-bbs was hälst du denn vor dem Vorschlag ein paar Zeilen oben drüber wo die dmsg genommen wird von 
`$dmsg = urlEncode(@{$ProtocolListRead}[$i]->{data}[$i2]->{$key}) if ($key =~ /dmsg/);`

>Die Zeile "$buttons.=........." ist schon ein arges Gewurschtel :-)

Das ist daraus resultierend weil auch du es sehr komfortabel haben wolltest *smile*

Natürlich ist auch die Variante von @sidey79 möglich.

elektron-bbs, 2020-05-13:
Das müsste eigentlich auch funktionieren.

HomeAutoUser, 2020-05-14:
Ich habe es geändert und hochgeschoben nach der https://github.com/RFD-FHEM/SIGNALduino_TOOL/issues/42#issuecomment-627881837 Variante.

Ob ich alles so umstelle wie https://github.com/RFD-FHEM/SIGNALduino_TOOL/issues/42#issuecomment-626948385, habe ich erstmal offen gelassen.

HomeAutoUser, 2020-12-29:
Mit der letzten Version habe ich den Punkt getestet stichprobenartig und die DMSG wurde dispatcht aus der geladenen JSON Datei.

closed 2020-12-29 by HomeAutoUser