RFD-FHEM/SIGNALduino_TOOL #44: Malformed MC Data in SD_Device_ProtocolList.json

sidey79, 2020-12-29:
In der SD_Device_ProtocolList.json sind fehlerhafte MC Daten, die aus irgendeiner Debug Firmware stammen.

Diese werden nicht mehr verarbeitet, wenn Parse_MC auf valide Datensätze z.B. `  if ($rmsg !~ /^M[cC];LL=-\d+;LH=\d+;SL=-\d+;SH=\d+;D=[0-9A-F]+;C=\d+;L=\d+;(?:R=\d+;)?$/){` prüft .

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/568e64fccf101a9d9e3c3b98f6ffc9da121abc88/FHEM/lib/SD_Device_ProtocolList.json#L295

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/568e64fccf101a9d9e3c3b98f6ffc9da121abc88/FHEM/lib/SD_Device_ProtocolList.json#L304

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/568e64fccf101a9d9e3c3b98f6ffc9da121abc88/FHEM/lib/SD_Device_ProtocolList.json#L427

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/568e64fccf101a9d9e3c3b98f6ffc9da121abc88/FHEM/lib/SD_Device_ProtocolList.json#L730

SD_Device_ProtocolList.json unterstützt aktuell keine fehlerhaften Daten. Um fehlerhafte Daten erkenne zu können müsste ein Flag eingebaut werden oder z.B. ein vorhandenes "dispatch_repeats":"-1" gesetzt werden.

HomeAutoUser, 2020-12-30:
Waren die Fehler, welche du hier mitbekommen hast für mich sichtbar oder hast du sie nur mitbekommen weil du „extern“ dies irgendwo getestet hast?

Ich überlege schon die ganze Zeit / nach einer Variante um die Daten stets plausibel zu halten ohne immer jede ID manuell zu dispatchen.

Das mit dem dispatch_repeats -1 habe ich nicht verstanden aber ich schau auch gerade nur mit dem Handy nach. Das muss ich mir am PC mal genau ansehen.

sidey79, 2020-12-30:
Die Daten werden durch das UnitTest Modul bei jedem push getestet:

https://github.com/RFD-FHEM/RFFHEM/runs/1623409933?check_suite_focus=true#step:12:4279

Ich sollte es geschwind in die neuere FHEM Testumgebung migrieren, dann liefert prove das Ergebnis samt möglicher Warnings.

HomeAutoUser, 2021-01-08:
Ich habe mir soeben die Fehler angesehen und auch korrigiert.
Nun sollte die Richtigkeit gegeben sein @sidey79.

sidey79, 2021-01-08:
Test läuft damit nun durch. 👍 
Das bedeutet aber auch, dass diese Firmware Versionen nicht kompatibel wären mit der Anpassung

closed 2021-01-08 by sidey79