RFD-FHEM/SIGNALduino_TOOL #50: Viele falsche RMSG

sidey79, 2021-02-13:
@HomeAutoUser 

Wenn wir die Prüfung der RMSGs wie in https://github.com/RFD-FHEM/RFFHEM/pull/926 angedacht machen, dann sind ganz viele rmsgs im JSON ungültig.

Z.B. diese hier:

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/67013e96de0a18f8cbbd64d9259218122efaa568/FHEM/lib/SD_Device_ProtocolList.json#L18

Das s4 wird in der Firmware nicht gesetzt. Ich kann mich da auch an nichts erinnern.

Bei MUs ist ein e drinnen, das ist mir auch unbekannt:
https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/67013e96de0a18f8cbbd64d9259218122efaa568/FHEM/lib/SD_Device_ProtocolList.json#L271

Insgesamt sind es 35 ungültige MS Nachrichten und und 15 MU (prerelase und master zusammen gerechnet).

HomeAutoUser, 2021-02-13:
@sidey79 die Differenzen sind von den beiden Firmwareversionen. Ich werde versuchen mir es anzusehen.

Was oder wie lässt du den Test ablaufen das er die RAWMSG als nicht plausibel aussonderst ? Durch ein RegEx?

sidey79, 2021-02-13:
Vermutlich kommt es davon ja.
Die regex sind die, welche wir ja entwickelt hatten. 
Ist halt die Frage, wie streng die Prüfung sein soll.
Wir könnten am Ende ja alle möglichen Zeichen erlauben, denn das im Blick halten was alles vorkommen kann finde ich nicht so prickelnd.
Die Prüfung deaktivierbar machen halte ich auch für keine alltags taugliche Lösung.

elektron-bbs, 2021-02-14:
Die Tests sollten doch wohl nicht strenger sein wie ein funktionierendes FHEM. Solange zusätzliche Variablen in den Nachrichten die Weiterverarbeitung im laufenden FHEM nicht stören, sollten sich auch die Tests nicht daran stören.
Oder willst du zukünftig Nachrichten vom MapleSduino generell ausschließen?

sidey79, 2021-02-14:
@elektron-bbs 

Den Ursprung nahm die Sache ja, dass die Prüfung auf valide Daten zu schwach ist und es zu warnings kommt.
Dann hatte @HomeAutoUser die Idee, dass mittels einer Regex zu prüfen und was nicht passt wird erst gar nicht angerührt.
Jetzt haben wir Tests mit Fehlerhaften Daten und eine Regex die sich an dem Datenformat aus https://github.com/RFD-FHEM/SIGNALDuino orientiert.  Andere Datenformate kenne ich erst mal nicht und wüsste auch nicht, wie die zu interpretieren wären. Aufgefallen ist es halt dadurch, dass im json wohl Daten stammen nicht nicht mit der Firmware aus https://github.com/RFD-FHEM/SIGNALDuino stammen.
In wie weit das nun Auswirkungen auf FHEM hat oder nicht, dazu kann ich überhaupt keine Aussage treffen.

Ich weiss auch selbst nicht was ich will.
Wenn wir die Regexprüfung weglassen, riskieren wir wieder "Müll" zu verarbeiten und es kommt zu warnings
Wenn wir unbekanntes in der regex erlauben wollen, wüsste ich gerade nicht wie, da ich noch nicht mal die Positionierung innerhalb des Datenstrings kenne. 
Irgendwie eine Zwickmühle.

elektron-bbs, 2021-02-14:
Ich würde nur prüfen, was wir benötigen und zusätzliche Variablen, sofern sie nicht den Ablauf stören, einfach ignorieren.

sidey79, 2021-02-14:
@elektron-bbs 

Verstehe ich Recht, dass Du auf eine Regex Prüfung verzichten würdest, welche die Bestandteile der rmsg auf valide Werte prüft?

elektron-bbs, 2021-02-14:
Natürlich nicht komplett auf die Prüfung verzichten, sondern auf relevante Bestandteile beschränken.
Z.B. stören doch diese Bestandteile "O;s=4;m1;" die weitere Verarbeitung in FHEM nicht. Also muss auch der Test diese nicht bemängeln.

sidey79, 2021-02-14:
m1 ist ja bekannt
O ist ebenfalls bekannt

s=4 ist ein unbekannter Wert.

Ich habe im Moment folgende Regex:
`^MS;(?:P[0-7]=-?\d+;){3,8}D=[0-7]+;CP=[0-9];SP=[0-9];(?:R=\d+;)?(?:O;)?(?:m=?[0-9];)?$`

Ich hab schon mal versucht mittels oder groups auch unbekanntes zu erlauben, das klappte bislang noch nicht:
`^MS;(?:P[0-7]=-?\d+;){3,8}D=[0-7]+;CP=[0-9];SP=[0-9];((?:R=\d+;)?|(?:O;)?|(?:m=?[0-9];)?|(?:[A-Za-z0-9]+;)+)$`

Das ist ja nun quasi das Problem, etwas erlauben das unbekannt ist, aber in den bekannten Feldern auch nur valides zulassen.

sidey79, 2021-02-14:
So scheint es auf den 1. Blick zu klappen. Ganz wohl ist mir dabei nicht gerade ;)

`^MS;(?:P[0-7]=-?\d+;){3,8}D=[0-7]+;CP=[0-9];SP=[0-9];((?:R=\d+;)?|(?:O;)?|(?:m=?[0-9];)?|(?:[A-Za-z0-9=]+;)*)$`

elektron-bbs, 2021-02-14:
Bei CP= und SP= müsste auch [0-7] reichen.

sidey79, 2021-02-14:
@elektron-bbs 
@HomeAutoUser 

Ich habe die regex angepasst:

https://github.com/RFD-FHEM/RFFHEM/pull/926/commits/5023de8aa4e6ddb3f3ea7eeb3c23407c889b94e6

passt noch nicht 100% aber es wird

sidey79, 2021-02-14:
@HomeAutoUser 
@elektron-bbs 

Ich hab nun genau den Fall, den ich nicht gefixt bekomme:

https://regex101.com/r/oQBkKv/1

Mach ich es zu streng, sind beide nicht erlaubt, mache ich es zu lose, dann werden beide verarbeitet, obwohl das ja dann nicht korrekt ist.