RFD-FHEM/SIGNALduino_TOOL #93: set command ProtocolList_save_to_file missing

sidey79, 2023-01-07:
Ich glaube der String wird wie folgt geschrieben `RAMSG´s are dispatched` 

https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/a97c673ebf75d3ddd02382e53711272185d00202/FHEM/88_SIGNALduino_TOOL.pm#L413

Zumindest steht es bei mir so und das Set Kommando fehlt.

HomeAutoUser, 2023-01-16:
Hallo @sidey79,
du wurdest nicht vergessen.
Bin ab und zu nur ohne Technik.

>Ich glaube der String wird wie folgt geschrieben RAMSG´s are dispatched

Geht es dir nur um die Schreibweise oder weil das Kommando nicht verfügbar ist?

>Zumindest steht es bei mir so und das Set Kommando fehlt.

Das Set Kommando ist nur verfügbar, wenn man auch Daten vorher geladen hat.
https://github.com/RFD-FHEM/SIGNALduino_TOOL/blob/master/FHEM/88_SIGNALduino_TOOL.pm#L4608:L4609

sidey79, 2023-01-16:
Ich hab über die Input Datei dispatched, aber der Befehl kam nicht. Er kommt nur, wenn eine einzelne Nachricht dispatched wird.

HomeAutoUser, 2023-01-16:
Ich denke zu wissen wie du meinst. Kannst du bitte denn noch mal die Schritte kurz schildern.

sidey79, 2023-01-20:
Ja, habe es jetzt wieder angewendet:

1. Attribut `File_input` auf eine Datei mit MU Messages angepasst
2. Attribut `File_input_StartString` auf MU angepasst.
3. `set Dispatch_file` aufgerufen
4. STATE ist anschließend = "finished, all RAMSG´s are dispatched (0.0739660263061523 second)"

Ich bekomme keinen set Befehl `ProtocolList_save_to_file`

HomeAutoUser, 2023-01-23:
Wenn ich das richtig deute, so suchst du den Set Befehl `ProtocolList_save_to_file` nachdem du alle MU Messages aus einer Datei dispatcht hast.

Für den Fall ist das nicht vorgesehen. Es wird nur beim manuellen dispatchen von einzelnen Nachrichten hinzugefügt, wenn man die Liste geladen hatte.

Ich muss mich mal versuchen in dein Vorhaben zu denken. Was möchtest du erzielen oder war dein Gedanke dabei?

sidey79, 2023-01-23:
Ich habe gedacht, ich gebe dem Tool das Logfile, lasse es Dispatchen und daraus wird mir dann ein JSON für den Test generiert. :(

HomeAutoUser, 2023-01-23:
Das ist schon wieder die Stufe weiter höher.

Da nach jedem angelegten Device die Werte geprüft werden und man diese separat anhaken kann / muss, so wird das in dem Fall nicht gemacht. Nur wenn einzeln dispatched wird.

Ich werde mit deinen Wunsch mal notieren ;-) aber bisher muss ich noch einen Fehler suchen im TOOL um testDaten zu genieren. ;-) -> wenn man diese einzeln dispatched

HomeAutoUser, 2023-01-25:
> Ich habe gedacht, ich gebe dem Tool das Logfile, lasse es Dispatchen und daraus wird mir dann ein JSON für den Test generiert. :(

@sidey79 was hat das für einen Sinn? Ich hoffe du meintest nicht die JSON Datei als Input nutzen zu wollen ;-)

sidey79, 2023-01-25:
> @sidey79 was hat das für einen Sinn? Ich hoffe du meintest nicht die JSON Datei als Input nutzen zu wollen ;-)

Nein JSON hatte ich als Output erwartet.
Mir ist halt so manches schleierhaft. :)

Folgende Situation:
Jemand erstellt ein Logfile mit RMSGs.
Diese Datei bringe ich auf den FHEM Host in dem auch das Tool läuft (Anmerkung, eine Texteingabe oder URL Angabe wäre einfacher)

Dann lasse ich das Tool verschiedene Dinge mit den RMSGs aus dem Log machen und das ganze auch Dispatchen.

Letztenendes hätte ich die Daten dann gerne in den Test eingebaut. Anstatt das JSON durch abtippen von Hand zu füllen, dachte ich das Tool nimmt mir hier einen Teil meiner Arbeit ab und erstellt die Datensätze. Ggf. Muss ich die halt noch anpassen.

Irgendwie kann man doch einen Testdatensatz mit dem Tool erzeugen. Das

HomeAutoUser, 2023-01-25:
>Mir ist halt so manches schleierhaft. :)

Verstehe ich, da es noch Neuland für dich ist.

>Folgende Situation:
Jemand erstellt ein Logfile mit RMSGs.

Das unterstützt es ja, das alles Dispatched wird.

>Letztenendes hätte ich die Daten dann gerne in den Test eingebaut. Anstatt das JSON durch abtippen von Hand zu füllen, dachte ich das Tool nimmt mir hier einen Teil meiner Arbeit ab und erstellt die Datensätze. Ggf. Muss ich die halt noch anpassen.

Das ganze wird derzeit unterstützt ABER nur wenn man die Nachrichten einzeln dispacht. Wenn man ein Device oder Protokoll entwickelt, so kommt das meist nur einzeln vor. Nach der Übernahme der Daten ins JSON werden dann auch TestDaten erzeugt. (Daran feile ich gerade noch, da @elektron-bbs und mir, deine ganzen Tests zu viel wurden in der Schreiberei ;-) )

Wenn der Fehler korrigiert ist im TOOL, kann ich Resultate zeigen.