RFD-FHEM/SIGNALduino_TOOL #99: get <sdtool'

sidey79, 2023-04-05:


sidey79, 2023-04-05:
Es wäre sinnvoll, wenn ein Filelog oder ein Link auf die Logdatei erzeugt werden würde um einfach über das WebFrontend an das gefilterte Log zu gelangen

HomeAutoUser, 2026-10-29:
Meinst du mit Logfile ein bestimmtes oder bei der Scan Funktion wie hier https://github.com/RFD-FHEM/SIGNALduino_TOOL/commit/7795135119ba8d4b778e9c5310a200f254d77a78 vollendet.

sidey79, 2026-10-30:
Ich kann mich nicht mehr erinnern.

Die Lösung mit dem Scanlog ist gut.
Es gab doch aber auch die Möglichkeit Daten aus einem Log zu filtern.

Liegen die gefilterten Daten dann im Dateisystem und der Zugriff aus FHEM heraus ist nicht möglich?

HomeAutoUser, 2026-11-02:
>         Die Lösung mit dem Scanlog ist gut.

Danke, mir war es schon immer ein Dorn im Auge, auf fremden Systemen nur über das Dateisystem ins Logfile zu schauen.

>         Es gab doch aber auch die Möglichkeit Daten aus einem Log zu filtern.
>         Liegen die gefilterten Daten dann im Dateisystem und der Zugriff aus FHEM heraus ist nicht möglich?

Die Möglichkeit gibt es, wenn man das Attribute File_input gesetzt ist.
Die Inputdatei muss im Ordner (Attribut Path) hinterlegt sein. Diese kann gefiltert werden mit dem get Befehlt "FilterFile". Das Resultat liegt im selben Ordner in der Datei des Attributes File_export.